luni64/TeensyTimerTool #2: Tick timer not working on Teensy LC

RocketManRC, 2020-01-28:
I really like the idea of this library and for my application the tick timer will be perfect. Unless I am doing something wrong I don't think it works on the Teency LC and I believe it is because ARM_DWT_CYCCNT is always 0 on that processor. To test I am using the example HelloOneShot.

Cheers!

luni64, 2020-01-28:
That's true, I started working around that a couple of days ago but didn't find time yet. In case you want to experiment, the systick counter might help: https://forum.pjrc.com/threads/28407-Teensyduino-access-to-counting-cpu-cycles?p=116317&viewfull=1#post116317
might do the trick

RocketManRC, 2020-01-28:
Yes I already use that counter directly in my application. It is only 16 bits so I don't think that will work in your library will it? I think you could just use micros() in your library for the Teensy LC and that would at least get it working. It would do the job for me at least.

I'm regretting using the Teensy LC but I had a few of them kicking around. That's never the best way to choose a microcontroller for a project :-)

luni64, 2020-01-28:
>Yes I already use that counter directly in my application. It is only 16 bits so I don't think that will work in your library will it? 

Sure, but extending to 32bit should not be too difficult...

> I think you could just use micros() in your library for the Teensy LC and that would at least get it working. It would do the job for me at least.

Good idea for a quick fix. Implemented and uploaded it. Can you give it a try?

>I'm regretting using the Teensy LC but I had a few of them kicking around. That's never the best way to choose a microcontroller for a project :-)

Why not? At least it triggered me fixing the issue :-)

RocketManRC, 2020-01-28:
That works great now and makes my code a lot cleaner and easier to understand. I can see that your library would be really beneficial once you need several timers and different types of timers. Very cool!

Thanks!

closed 2020-02-03 by luni64