luni64/TeensyTimerTool #3: Installation instructions.

h5n1xp, 2020-03-01:
There needs to be detailed instructions as to have this library can be integrated into the Arduino software.

luni64, 2020-03-01:
Thanks, I'll add something to the readme. 

Generally this library can be installed like any other Arduino library.  Here some info about the process  https://forum.pjrc.com/threads/59112-TeensyTimerTool?p=231529&viewfull=1#post231529

h5n1xp, 2020-03-01:
Fantastic, this is more of an issue with the Arduino software than with TeensyTimerTool.

I’ve been follow your development thread on the forum, it looks to me like exactly what the Teensy platform is missing. The current ARM chips have so much peripheral  hardware, and much of it inaccessible with with the Arduino software.

closed 2020-08-28 by luni64