luni64/TeensyTimerTool #5: setPeriod() on PeriodicTimer / Direct usage of ITimerChannel

imwhocodes, 2020-06-09:
Hi,

Is possible to change the timer period on the fly?

I would like to change the period of the timer without calling `timer.begin(...)` again.

Let's say that I have a `PeriodicTimer` running and I want change its period without resetting its internal counter of time already passed

Searching through the source code I found that `ITimerChannel` has some extra/usefull member function like `setPeriod` or `setCallback` but i'm unable to find an actual implementation of the aforementioned class even if it is an attribute of `PeriodicTimer`

Can I subclass `PeriodicTimer` to expose `timerChannel->setPeriod` and `timerChannel->setPeriod`? Is this safe? or changing  period to a running timer is really unsafe / not doable?

Thanks,
Luca

luni64, 2020-06-09:
I planned to provide that functionality (hence the stubs) but never found time to actually implement it. 
Subclassing would work but please not that the feature currently is only implemented for the TCK timers. 

Instead of subclassing you could activate `ENABLE_ADVANCED_FEATURES` in the config file. And use the then available getChannel() function of the timer. 
https://github.com/luni64/TeensyTimerTool/blob/abea02d990115e17e48a82576b16b33a003a40ba/src/baseTimer.h#L11-L30

It returns a pointer to the allocated timer channel. Just cast it to the correct timer channel type and you can use the `setPeriod()` function of it. 

Let me know if you need a worked out example

imwhocodes, 2020-06-15:
Ok, In my spare time I started checking the T4.x IMXRT1060RM [datasheet](https://www.pjrc.com/teensy/IMXRT1060RM_rev2.pdf)

I just forked and tried to put down some code, [a pure WIP](https://github.com/imwhocodes/TeensyTimerTool), not even bothering compiling

The main theme is that I think the best approach is to split `setPeriod` in two function:
- `setCurrentPeriod`
- `setNextPeriod`

Where the first set the period for the current running counter, the latter set the period loaded in the counter/comparator after the current period triggered/ended

From the documentation seem like we can use this approach for all T4 timers with differents workaround

Maybe we can keep a `setPeriod` that configure both **`current`** and **`next`** simultaneously, so it will not even be a breaking change to the library

As i said the linked code is only to start messing around, at this moment far from a pull request, mostly to chose how to implement it

Luca

luni64, 2020-06-17:
Sounds good, I'll have a quick look at your code today

luni64, 2020-06-17:
Had a look at your code in master and develop. Looks good and you even cleaned up some of the mess I left the code in :-). Looking forward to do some tests. 

Unfortunately you decided to reformat and rearrange large parts of the code which makes it very difficult for me to follow your changes. I know that it is always tedious to follow the coding style of others but it would make my life much easier if I could see what was changed in the git history when it comes to merging. 

**The following is important to me:** 
- I don't like code in class declarations. (I only do this for quick hacks and often forget to remove it later :-) ).  If I do header only classes I usually define the functions in an implementation section below the definition. 

- I prefer having public declarations above private/protected ones. 

Both help me to quickly scan the functionality of a class without needing to wade through implementation details. 

Generally: It might be better to do more small pull requests instead of one large. So, if you have something working (maybe the TCK timer?) I'd be happy to pull it in and do some testing.

imwhocodes, 2020-06-17:
Very good

We’re on the same line on the header implementation separation, I actually tried to put the implementation on a cpp file but was to much at this early stage

I refactored almost everything starting again from your master branch, dividing on small valid (compilable) commits, everything is on the `development-cleaner-commit` branch

By now I implemented it for the T4.x and T3.x TCK timer and T4.x QUAD timer, but the branch is still a WIP so no pull request yet (actually I disable setPeriod I’m the other timers for refactoring purpose)

Yesterday I tried the QUAD timer of the `develop` branch with the oscilloscopes and the various `set*Period(uint32_t us)` seem to work correctly but I didn’t test yet the new branch

Luca

luni64, 2020-06-21:
I had a look at your _development-cleaner-commit_ branch. This does all look very good to me. Since I need to do some work on the stopping of the timers I pulled your commits into my branch _feature/stopTimer_ and work and test from this.

timidtapir, 2021-01-23:
I'm late to this dance, but it seems `getChannel()` was removed as part of the Oct 22, 2020 commit (!?)

Is there a new way of changing the period of a timer that is being worked on?   (I've simply added `getChannel()` back into my local, but that puts me out of sync and presumably more changes are on the way (?) )  

FWIW I'm primarily interested in `setNextPeriod()` type functionality ...

luni64, 2021-01-24:
I removed getChannel to clean up and because my impression was that it is too complicated to use. (need to cast to the correct channel type...). If you actually use it I can implement it back in a next release. 

> FWIW I'm primarily interested in setNextPeriod() type functionality ...

For which timer do you need that?

timidtapir, 2021-01-24:
Well, I just want to be able to change the period, and it seemed like `getChannel()` enabled that.  It would seem even nicer if there were method(s) up in `PeriodicTimer` (even if they're "hidden" behind an `ENABLE_ADVANCED_FEATURES` or equivalent) so less of the internal implementation needed to be exposed.

To your question, I'm using TMR's, but solely because that was what you supported changing period on (and the shorter period is adequate).  Earlier I was using `IntervalTimer`s and the `update` method (i.e. PIT) which also worked fine.   BTW I'm changing the period continuously after each cycle, which enables some fun tricks,

mateusz-kusmierz, 2021-06-11:
Hello to everyone!
I'm using PeriodicTimer in my project, and I also need to change it's period on the fly.
As i understand, this feature is absent in current master code?

luni64, 2021-06-14:
I implemented setPeriod and setNextPeriod in the development branch for the PIT timers. Can you give it a try?

closed 2021-07-19 by luni64