luni64/TeensyTimerTool #7: Protect unsafe usage of `BaseTimer::start` and `BaseTimer::stop`

korhadris, 2021-01-02:
I know I shouldn't be doing this, but I noticed that `BaseTimer::start` and `BaseTimer::stop` use the `timerChannel` pointer without checking them; I would expect them to return `errorCode::notInitialized` if it isn't set yet.

Test to reproduce crashes:
```c++
#include <TeensyTimerTool.h>

TeensyTimerTool::PeriodicTimer t1;

void begin() {
  t1.start();  // Dies here without progressing (expected seg fault)
  // Rest of code
}
```

korhadris, 2021-01-02:
I don't mind making a PR for this change if wanted too...

luni64, 2021-01-03:
Thanks for spotting this. I'll fix it later today

luni64, 2021-11-08:
Done in https://github.com/luni64/TeensyTimerTool/commit/7eab51926f3d5ef96103bef7fd1741c6f6b7415f

closed 2021-11-08 by luni64