luni64/TeensyTimerTool #8: Including both TeensyTimerTool and TimeLib.h

puregame, 2021-01-10:
I'm playing around with a T4.1 and I want to use the TeensyTimerTool in a project that also will be using the RTC functionality.

I've narrowed down an issue to when both <TimeLib.h> and "TeensyTimerTool.h" are both #included in the project. I tested this with an empty project and get compile issues. Any help on how to get past these compile issues would be great. I can't find anything similar on google.

[image]

luni64, 2021-01-11:
Looks like TimeLib.h messes up the standard <chrono> time facility of c++ 11.  If you replace the TeensyTimerTool include by include <chrono> you'll get the same issue which of course is a pitty but I can't do much about it. 
You can disable the use of std::chrono in the TeensyTimerTool config  (https://github.com/luni64/TeensyTimerTool/blob/77c116db1dc29077325b03f9a81684da45b72855/src/defaultConfig.h#L80) to work around this. This will of course disable the use of the time literals, i.e. you can't do things  like `trigger(25ms)` anymore.  (https://github.com/luni64/TeensyTimerTool/wiki/Configuration)

puregame, 2021-01-12:
That worked, I had tried removing `Time.h` from the folder but that alone did not work. I also renamed `Time.cpp` to `TimeLib.cpp`. All code still compiles and works as intended.

Thank you!

closed 2021-01-12 by puregame

MaffooClock, 2021-03-02:
Confirmed: in the Time library, delete Time.h and rename Time.cpp to TimeLib.cpp