luni64/TeensyTimerTool #10: [Help] Using timer ticks instead of microseconds ?

MaximeGregot, 2021-02-19:
Hi,

_[sorry if this is not the correct place to post this message, I don't know where else to put it]_

I'm a beginner on the Teensy 4.1 microcontroller, and I was very relieved to see that this library exists (so thanks a lot !).
However in order to control several stepper motors with high speeds, using only microseconds as timer intervals isn't very handy for my use of the library.
So is there a way to use nanoseconds, or even timer ticks (which would be ideal), with the general purpose timers `timer.trigger(...)` function ?

It would be very handy if I could simply set a 32-bit number, at which the timer will trigger the interrupt callback function.
I hope that my question was clear...

Thanks to anyone who will take time to answer,
Maxime.

luni64, 2021-02-20:
The library takes floats for the period inputs. So, just use 0.5 if you need 500ns. Does that help?

The timer tool doesn't take counter reloads as input. Would be quite against the intention to provide a simple interface to the timers.  Actually, I can't imagine a situation where a counter reload value would be more convenient than a time value?

MaximeGregot, 2021-02-20:
Oh right... I was almost sure that it required an integer, as all the examples in the wiki were with integers.
So yes, a big yes, that helps ! Sorry for asking what ends up being a dumb question, and thanks for answering so quickly !

closed 2021-02-23 by luni64