luni64/TeensyTimerTool #12: Adjust FTM Timer Interval in Real Time?

djex, 2021-04-08:
I need to be able to adjust a FTM timers interval in real time while it is running. I can not find a way to do this. Is this possible? If so how would I go about doing this? I've tried using a one shot timer that triggers it self though a call back using a variable for the delay but when the delay is changed it does not update until the timer is stopped and started again. Not sure why.

luni64, 2021-04-08:
Unfortunately an a update of the period of periodic timers is currently not implemented (but on my todo list). It should work with one shot timers if you don't require super accurate results. Can you post some (minimal) code showing the issue with the one shot timers?

djex, 2021-04-08:
I may have a look into your code and implement it myself. I need fairly decent accuracy as I'm using this as a MIDI clock source. However since you asked for the code using the one shot timer I included a simplified version of what I have below. If I call UpdateDelay() while the one shot loop is running the delay never changes. However if I stop the loop by setting clockStarted to false and then start it again the new delay will take effect. It's like the one shot timer is locking the delay variable and only allowing reads while it is active.

#include <TeensyTimerTool.h>

using namespace TeensyTimerTool;
OneShotTimer timerInternalMIDIClockOneShot(FTM0);

double delay = 200;

bool clockStatred = false;
void TimerOneShot()
{
  if (clockStatred)
  {
    //do stuff here

    timerInternalMIDIClockOneShot.trigger(delay);
  }
}

void ToggleInternalMIDIClock() 
{
    if (clockStatred)
    {
        clockStatred = false;
    }
    else
    {
        clockStatred = true;;
        TimerOneShot();
    }
}

void UpdateDelay(double newDelay)
{
    delay = newDelay;
}

void setup()
{
    timerInternalMIDIClockOneShot.begin(TimerOneShot);
}

luni64, 2021-04-08:
Unfortunately your code is not compiling. Loop is missing and 'delay' is not a good name since it clashes with the built in 'delay()' function. 
Here and example which shows how to use one shot timers to change speed on the fly. I change the delay from 1ms to 2ms every 100ms. 

```c++
#include "Arduino.h"
#include <TeensyTimerTool.h>

using namespace TeensyTimerTool;
OneShotTimer timerInternalMIDIClockOneShot(FTM0);

unsigned currentDelay = 1'000;

bool clockStarted = false;
void TimerOneShot()
{
    if (clockStarted)
    {
        digitalWriteFast(0, HIGH); // marker for LA
        delayMicroseconds(1);
        digitalWriteFast(0, LOW);

        timerInternalMIDIClockOneShot.trigger(currentDelay);
    }
}

void ToggleInternalMIDIClock()
{
    if (clockStarted)
    {
        clockStarted = false;
    } else
    {
        clockStarted = true;
        TimerOneShot();
    }
}

void setup()
{
    for (int pin : {LED_BUILTIN, 0, 1, 2})
    {
        pinMode(pin, OUTPUT);
    }

    timerInternalMIDIClockOneShot.begin(TimerOneShot);
    ToggleInternalMIDIClock();
}

void loop()
{
    delay(100);
    currentDelay = (currentDelay == 1'000) ? 2'000 : 1'000; // toggle delay betwend 1ms and 2ms every 100ms

    digitalWriteFast(1, HIGH); // generate marker for LA
    delayMicroseconds(1);
    digitalWriteFast(1, LOW);

    digitalToggleFast(LED_BUILTIN); // heartbeat
}
```

Here the output of pin 0 and pin 1:

[image]

Zoomed in: 
[image]

djex, 2021-04-08:
Your code works perfect and it allowed me to find the issue with my code. Thank you. However I've noticed something strange. A delay of anything over 17605 does not work. The timing becomes very unstable and erratic.I'm guessing this is a limitation of the FTM timer?

luni64, 2021-04-09:
I'm a bit confused. The FTMs are 16bit timers. They are clocked by F_BUS (default: 60MHz and 1/32 prescaler). That gives a max period of  2^16 * 32/60E6 = 34.95ms.  If you select anything higher than this it will be set to 34.95ms.  You can easily query the max period of a timer by getMaxPeriod(). Here the corresponding chapter in the documentation:  https://github.com/luni64/TeensyTimerTool/wiki/Basic-Usage#finding-out-the-maximum-timer-period

Also, I strongly suggest to use the error handling capabilities of the lib during development. Here a slightly changed `setup()`

```c++
void setup()
{
    while(!Serial){}
    TeensyTimerTool::attachErrFunc(ErrorHandler(Serial));

    for (int pin : {LED_BUILTIN, 0, 1, 2})
    {
        pinMode(pin, OUTPUT);
    }

    timerInternalMIDIClockOneShot.begin(TimerOneShot);

    Serial.printf("FBUS: %d, max period: %.3f ms\n",F_BUS, timerInternalMIDIClockOneShot.getMaxPeriod()*1000);

    ToggleInternalMIDIClock();
}
```

Output: 
[image]

If you select a period larger than the max period by e.g. adding `timerInternalMIDIClockOneShot.trigger(200'000);`  to `setup()` you'll get a warning: 

[image]

djex, 2021-04-10:
Thanks. Yes I am getting a "W-100: Period overflow, set to maximum" .This explains my issue. Thank you. I think I'm just going to have to settle for using the IntervalTimer.

closed 2021-07-19 by luni64