luni64/TeensyTimerTool #15: Destructor on TCK timer

carlkesselman, 2021-09-10:
I'm using a one shot TCK64 timer on Teensy 4.1.  I am creating and deleting timers.  However, deleting timers doesn't seem to work in that timers don't get released.

It looks like the destructor for TCK timers is defined, but has an empty body.  I have also seen that there is a function called removeTImer, but this is never used.  Is there any reason why the removeTImer function could not be used in the destructor so that timer slots could be reused.

Thanks,

Carl

luni64, 2021-09-10:
I never bothered to do proper destructors since Arduino folks mostly create such things globally and objects usually live forever...
Anyway, that certainly needs to be fixed and I'll take care of it asap.

luni64, 2021-09-10:
Can you give the code in the development branch a try?

carlkesselman, 2021-09-10:
Nice.  That seems to work.

Carl

----------------------------------------------------------

Dr. Carl Kesselman
William H. Keck Professor of Engineering
Professor, Epstein Department of Industrial and Systems Engineering
Professor, Ostrow School of Dentistry
Professor, Preventive Medicine Keck School of Medicine
Fellow, Information Sciences Institute
Viterbi School of Engineering

University of Southern California
4676 Admiralty Way, Suite 1001, Marina del Rey, CA  90292-6695
Phone:  +1 (310) 448-9338
Email:  ***@***.***
Web: http://www.isi.edu/~carl

luni64, 2021-09-10:
Good, I'll add the missing ones (PIT, FTM), do some tests and make a new release the next days

luni64, 2021-11-07:
Done in v0.4.1

closed 2021-11-07 by luni64