luni64/TeensyTimerTool #17: Variable One-Shot Delays (broken?)

PowerBroker2, 2021-12-18:
This looks like a very promising library to help with a PAL video encoder project I'm working on, but it seems that the TeensyTimerTool isn't letting me use multiple timers or updating the timers I'm using. For example:

```C++
#include "TeensyTimerTool.h"

using namespace TeensyTimerTool;

OneShotTimer timer1;
OneShotTimer timer2;

byte ledState = HIGH;

void processVideo()
{
  if (ledState == HIGH)
  {
    ledState = LOW;
    timer2.trigger(2000);
  }
  else
  {
    ledState = HIGH;
    timer1.trigger(1000);
  }
  
  digitalWriteFast(LED_BUILTIN, ledState);
}

void setup()
{
  Serial.begin(115200);
  
  pinMode(LED_BUILTIN, OUTPUT);

  timer1.begin(processVideo);
  timer2.begin(processVideo);
  timer1.trigger(1000);
}

void loop()
{
  
}
```

From my understanding of the library, the above sketch should light the Teensy 4.0's built-in LED for 2 seconds, turn off for 1 second, and repeat. However, the LED just lights and stays lit.

Can you confirm this as a bug or am I misunderstanding how the library works?

PowerBroker2, 2021-12-19:
Nvm, I got it to work with a different library.

closed 2021-12-19 by PowerBroker2

luni64, 2021-12-19:
To avoid confusion for readers who might stumble over this later:
 
- In your code you set the time to 1 and 2 ms instead of sec. The timers expect microseconds not milliseconds as input.  
- Please also note that, if you don't specify the type of the timer in the constructor it uses the next free timer from the pool. In your case you get TMR4/channel1 and TMR4/channel2 both timers are 16bit wide and have a max duration of 2^16 /150MHz * 128 = 56ms. 
Here how to achieve your timings using the 32bit TCK timers and using time literals for defining the trigger time
```c++
#include "TeensyTimerTool.h"
using namespace TeensyTimerTool;

OneShotTimer timer1(TCK), timer2(TCK);

byte ledState = HIGH;

void processVideo()
{
    if (ledState == HIGH)
    {
        ledState = LOW;
        timer2.trigger(2s);
    }
    else
    {
        ledState = HIGH;
        timer1.trigger(1s);
    }
    digitalWriteFast(LED_BUILTIN, ledState);
}

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);

    timer1.begin(processVideo);
    timer2.begin(processVideo);
    timer1.trigger(1s);
}

void loop()
{
}
```
And here a more terse version of your code: 
```c++
#include "TeensyTimerTool.h"
using namespace TeensyTimerTool;

OneShotTimer timer1(TCK);

void processVideo()
{
    digitalToggleFast(LED_BUILTIN);
    timer1.trigger(digitalReadFast(LED_BUILTIN) ? 1s : 2s);
}

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
    timer1.begin(processVideo);
    timer1.trigger(10ms);
}

void loop(){
}
```

PowerBroker2, 2021-12-19:
Thank you! The example code works great. This is a really cool library, I just wish the documentation was better. Do you ever plan on making a sort of "read the docs" page for it?

luni64, 2021-12-20:
> I just wish the documentation was better. Do you ever plan on making a sort of "read the docs" page for it?

I agree that a library without documentation is quite useless. I therefore always try to document my stuff as good as possible. So, I don't know if you didn't find the documentation (https://github.com/luni64/TeensyTimerTool/wiki) or if you miss something in it?