luni64/TeensyTimerTool #18: v0.4.1 not available in the Arduino Library Manager

arnogramatke, 2022-02-11:
Hi,

the last release of the TeensyTimerTool that I can see in the Arduino Library Manager is v.0.4.0.

It looks like the order of tagging v0.4.1 and updating the library.json was mixed up. Would be great to have v0.4.1 available in the Library Manager.

Thanks
Arno

[image]

luni64, 2022-02-11:
Updated to 0.4.3. Lets hope this time the syncing works as expected

luni64, 2022-02-11:
Is updated: 
[image]

closed 2022-02-11 by luni64