luni64/TeensyTimerTool #21: Version 1.1.0 - Using timers in a class does not work

pawelplodzpl, 2022-09-26:
As of 1.1.0 the example code - How to Encapsulate a Timer and its Callback in a Class (https://github.com/luni64/TeensyTimerTool/wiki/Callbacks#how-to-encapsulate-a-timer-and-its-callback-in-a-class) no longer works.

Using lambdas inside classes while capturing [this] fails on latest Arduino IDE (2.0.0) and PlatformIO (v 2.5.4).

Problematic code:
```
void begin()  // better not initalizie peripherals in constructors
    {
       pinMode(pin, OUTPUT);
        timer.begin([this] { this->blink(); }, period);
   }
```

Full error log:

> in file included from c:\Users\pavel\Documents\Arduino\libraries\TeensyTimerTool\src/ITimerChannel.h:3:0,
                 from c:\Users\pavel\Documents\Arduino\libraries\TeensyTimerTool\src/API/baseTimer.h:5,
                 from c:\Users\pavel\Documents\Arduino\libraries\TeensyTimerTool\src/API/oneShotTimer.h:4,
                 from c:\Users\pavel\Documents\Arduino\libraries\TeensyTimerTool\src/TeensyTimerTool.h:3,
                 from D:\Codes\Arduino\code\timer\timertest\Blinker.hpp:2,
                 from D:\Codes\Arduino\code\timer\timertest\timertest.ino:1:
c:\Users\pavel\Documents\Arduino\libraries\TeensyTimerTool\src/types.h:9:10: warning: #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks" [-Wcpp]
         #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks"
          ^
In file included from D:\Codes\Arduino\code\timer\timertest\timertest.ino:1:0:
D:\Codes\Arduino\code\timer\timertest\Blinker.hpp: In member function 'void Blinker::begin()':
D:\Codes\Arduino\code\timer\timertest\Blinker.hpp:16:54: error: no matching function for call to 'TeensyTimerTool::PeriodicTimer::begin(Blinker::begin()::<lambda()>, unsigned int&)'
In file included from c:\Users\pavel\Documents\Arduino\libraries\TeensyTimerTool\src/API/oneShotTimer.h:4:0,
                 from c:\Users\pavel\Documents\Arduino\libraries\TeensyTimerTool\src/TeensyTimerTool.h:3,
                 from D:\Codes\Arduino\code\timer\timertest\Blinker.hpp:2,
                 from D:\Codes\Arduino\code\timer\timertest\timertest.ino:1:
c:\Users\pavel\Documents\Arduino\libraries\TeensyTimerTool\src/API/baseTimer.h:45:15: note: candidate: template<class period_t> TeensyTimerTool::errorCode TeensyTimerTool::BaseTimer::begin(TeensyTimerTool::callback_t, period_t, bool)
     errorCode BaseTimer::begin(callback_t callback, period_t p, bool start)
               ^
c:\Users\pavel\Documents\Arduino\libraries\TeensyTimerTool\src/API/baseTimer.h:45:15: note:   template argument deduction/substitution failed:
In file included from D:\Codes\Arduino\code\timer\timertest\timertest.ino:1:0:
D:\Codes\Arduino\code\timer\timertest\Blinker.hpp:16:54: note:   cannot convert '<lambda closure object>Blinker::begin()::<lambda()>{((Blinker*)this)}' (type 'Blinker::begin()::<lambda()>') to type 'TeensyTimerTool::callback_t {aka void (*)()}'
exit status 1
Compilation error: exit status 1

luni64, 2022-09-27:
> #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks"
Sorry, V1.1.0 needs the new toolchain (teensyduino beta1.58). Admittedly, it was not a good idea to switch to the new toolchain while it is still in beta. 
As a workaround you you can either use the beta toolchain or downgrade to 1.0.0 of the TimerTool. Changes from 1.0.0 to 1.1.0 are related to the new callback system which requires GCC  > V7.

bdlabitt, 2023-01-25:
I had this same problem with TTT.  IDE2.0.3.  Lots of errors.  Reverted to V1.0.0 and stuff worked again.  Relying on beta Teensyduino is a bad idea, since it is not shipped with the IDE.

luni64, 2023-01-25:
The current version is 1.2.0 which should fix the issue with the beta version. Please let me know if it doesn't

bdlabitt, 2023-01-25:
It didn't.  I installed Teensy 0.58.3.  I have V1.2.0 installed
`In file included from /home/bruce/Arduino/libraries/TeensyTimerTool/src/types.h:9,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/ITimerChannel.h:3,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/baseTimer.h:5,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/oneShotTimer.h:4,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/TeensyTimerTool.h:3,
                 from /home/bruce/Arduino/ELS_IDE2/ELS.h:15,
                 from /home/bruce/Arduino/ELS_IDE2/ELS_IDE2.ino:1:
/home/bruce/Arduino/libraries/TeensyTimerTool/src/inplace_function.h:10:13: error: redefinition of 'void panic()'
   10 | inline void panic() // could be weak to be overriden by a user error handler?
      |             ^~~~~

In file included from /home/bruce/Arduino/libraries/TeensyTimerTool/src/types.h:9,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/ITimerChannel.h:3,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/baseTimer.h:5,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/oneShotTimer.h:4,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/TeensyTimerTool.h:3,
                 from /home/bruce/Arduino/ELS_IDE2/ELS.h:15,
                 from /home/bruce/Arduino/ELS_IDE2/ELS_IDE2.ino:1:
/home/bruce/Arduino/libraries/TeensyTimerTool/src/inplace_function.h:100:33: error: redefinition of 'constexpr const size_t stdext::inplace_function_detail::InplaceFunctionDefaultCapacity'
  100 |         static constexpr size_t InplaceFunctionDefaultCapacity = 32;
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from /home/bruce/Arduino/libraries/TeensyTimerTool/src/types.h:9,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/ITimerChannel.h:3,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/baseTimer.h:5,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/oneShotTimer.h:4,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/TeensyTimerTool.h:3,
                 from /home/bruce/Arduino/ELS_IDE2/ELS.h:15,
                 from /home/bruce/Arduino/ELS_IDE2/ELS_IDE2.ino:1:
/home/bruce/Arduino/libraries/TeensyTimerTool/src/inplace_function.h:107:15: error: redefinition of 'union stdext::inplace_function_detail::aligned_storage_helper<Cap>'
  107 |         union aligned_storage_helper
      |               ^~~~~~~~~~~~~~~~~~~~~~

In file included from /home/bruce/Arduino/libraries/TeensyTimerTool/src/types.h:9,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/ITimerChannel.h:3,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/baseTimer.h:5,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/oneShotTimer.h:4,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/TeensyTimerTool.h:3,
                 from /home/bruce/Arduino/ELS_IDE2/ELS.h:15,
                 from /home/bruce/Arduino/ELS_IDE2/ELS_IDE2.ino:1:
/home/bruce/Arduino/libraries/TeensyTimerTool/src/inplace_function.h:131:16: error: redefinition of default argument for 'unsigned int Align'
  131 |         struct aligned_storage
      |                ^~~~~~~~~~~~~~~

In file included from /home/bruce/Arduino/libraries/TeensyTimerTool/src/types.h:9,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/ITimerChannel.h:3,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/baseTimer.h:5,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/oneShotTimer.h:4,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/TeensyTimerTool.h:3,
                 from /home/bruce/Arduino/ELS_IDE2/ELS.h:15,
                 from /home/bruce/Arduino/ELS_IDE2/ELS_IDE2.ino:1:
/home/bruce/Arduino/libraries/TeensyTimerTool/src/inplace_function.h:149:16: error: redefinition of 'struct stdext::inplace_function_detail::wrapper<T>'
  149 |         struct wrapper
      |                ^~~~~~~

In file included from /home/bruce/Arduino/libraries/TeensyTimerTool/src/types.h:9,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/ITimerChannel.h:3,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/baseTimer.h:5,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/oneShotTimer.h:4,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/TeensyTimerTool.h:3,
                 from /home/bruce/Arduino/ELS_IDE2/ELS.h:15,
                 from /home/bruce/Arduino/ELS_IDE2/ELS_IDE2.ino:1:
/home/bruce/Arduino/libraries/TeensyTimerTool/src/inplace_function.h:155:16: error: redefinition of 'struct stdext::inplace_function_detail::vtable<R, Args>'
  155 |         struct vtable
      |                ^~~~~~

In file included from /home/bruce/Arduino/libraries/TeensyTimerTool/src/types.h:9,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/ITimerChannel.h:3,
                 from /home/bruce/Arduino/libraries/TeensyTimerTool/src/API/baseTimer.h:5,
[... 43 lines cut ...]
      |     ~~~~~
  228 |     F,
      |     ~~  
  229 |     Args...> : std::true_type
      |     ~~~~~~~~

[... cut ...]
/home/bruce/Arduino/libraries/TeensyTimerTool/src/inplace_function.h:234:8: error: redefinition of 'struct stdext::inplace_function_detail::is_invocable_r_impl<decltype ((declval<F>()((declval<Args>)()...), void())), const void, F, Args ...>'
  234 | struct is_invocable_r_impl<
      |        ^~~~~~~~~~~~~~~~~~~~
  235 |     decltype(std::declval<F>()(std::declval<Args>()...), void()),
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  236 |     const void,
      |     ~~~~~~~~~~~
  237 |     F,
      |     ~~  
  238 |     Args...> : std::true_type
      |     ~~~~~~~~`
There's more, but you get the idea.

luni64, 2023-01-27:
see https://forum.pjrc.com/threads/70264-EncoderTool?p=319670&viewfull=1#post319670

closed 2023-01-27 by luni64