luni64/TeensyTimerTool #22: Teensy Timer Tool has 256 errors all of a sudden while using PlatformIO.

sbfreddie, 2022-11-15:
I have been using the Teensy Timer Tool for about a month in PlatformIO and never had any problems with it.  Now all of a sudden I get 265 strange errors.
Attached is the PIO built output. 
[PIO Error Output.txt](https://github.com/luni64/TeensyTimerTool/files/10016491/PIO.Error.Output.txt)

Any Suggestions?

Ed

luni64, 2022-11-16:
Looks like another issue due to the Arduino madness of #defining things like abs, min, max etc. This messes up parts of the STL if Arduino is included before STL headers. 
I can safeguard this. Can you post your project?

luni64, 2022-11-16:
Thanks for the code. I had a look and can reproduce the issue when I comment in the TimerTool include in TestingTFT.h.  

Workaround if needed:
Since the Intervaltimer works for you I assume that you neither use the modern callback system nor the std::chrono extenstion? In that case you can set the library to not use std::function and std::chrono in defaultConfig.h (line 76 and 84) 

Anyway, I'll use your code to track down the offending #include and fix it in the next release. Let me know if you need a fix more urgently despite the working intervaltimer solution.

sbfreddie, 2022-11-16:
Thanks, but the interval timer solution is fine for me as you can see that I am not doing anything fancy to generate a 1 second interrupt.  The interval timer solution has been in use for me for about 8-10 years continually on my Teensy 3.2 based clock-weather station.
If it would be possible could be please let me know what you do fix this problem when you get around to it.

Thanks,
Ed

closed 2023-08-14 by luni64