luni64/TeensyTimerTool #23: Problem with DoubleExposure example

mnissov, 2022-11-23:
I am trying to modify the DoubleExposure example as my use case is very similar, however I can't compile even the `PulseGenerator.h`

It returns the following errors
```
 *  Executing task: platformio run 

Processing teensy41 (platform: teensy; framework: arduino; board: teensy41)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.17.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.157.220801 (1.57) 
 - tool-teensy @ 1.157.0 (1.57) 
 - toolchain-gccarmnoneeabi @ 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 93 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TeensyTimerTool @ 1.1.0
Building in release mode
Compiling .pio/build/teensy41/src/with_library.cpp.o
Compiling .pio/build/teensy41/lib188/TeensyTimerTool/API/Timer.cpp.o
Compiling .pio/build/teensy41/lib188/TeensyTimerTool/API/baseTimer.cpp.o
Compiling .pio/build/teensy41/lib188/TeensyTimerTool/ErrorHandling/error_handler.cpp.o
Compiling .pio/build/teensy41/lib188/TeensyTimerTool/TimerModules/GPT/GPTChannel.cpp.o
Compiling .pio/build/teensy41/lib188/TeensyTimerTool/TimerModules/PIT4/PIT.cpp.o
Compiling .pio/build/teensy41/lib188/TeensyTimerTool/TimerModules/TCK/TCK.cpp.o
In file included from .pio/libdeps/teensy41/TeensyTimerTool/src/ITimerChannel.h:3:0,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/API/baseTimer.h:5,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/API/oneShotTimer.h:4,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/TeensyTimerTool.h:3,
                 from include/PulseGenerator.h:4,
                 from src/with_library.cpp:1:
.pio/libdeps/teensy41/TeensyTimerTool/src/types.h:9:10: warning: #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks" [-Wcpp]
         #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks"
          ^
Compiling .pio/build/teensy41/lib188/TeensyTimerTool/TimerModules/TCK/TckChannelBase.cpp.o
In file included from .pio/libdeps/teensy41/TeensyTimerTool/src/ITimerChannel.h:3:0,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/API/timer.h:3,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/API/Timer.cpp:1:
.pio/libdeps/teensy41/TeensyTimerTool/src/types.h:9:10: warning: #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks" [-Wcpp]
         #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks"
          ^
In file included from .pio/libdeps/teensy41/TeensyTimerTool/src/ITimerChannel.h:3:0,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/API/baseTimer.h:5,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/API/baseTimer.cpp:1:
.pio/libdeps/teensy41/TeensyTimerTool/src/types.h:9:10: warning: #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks" [-Wcpp]
         #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks"
          ^
In file included from .pio/libdeps/teensy41/TeensyTimerTool/src/ErrorHandling/error_handler.cpp:3:0:
.pio/libdeps/teensy41/TeensyTimerTool/src/types.h:9:10: warning: #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks" [-Wcpp]
         #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks"
          ^
Compiling .pio/build/teensy41/lib188/TeensyTimerTool/TimerModules/TCK/tickCounters.cpp.o
Compiling .pio/build/teensy41/lib188/TeensyTimerTool/config.cpp.o
In file included from .pio/libdeps/teensy41/TeensyTimerTool/src/ITimerChannel.h:3:0,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/TimerModules/GPT/GPTChannel.h:4,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/TimerModules/GPT/GPTChannel.cpp:3:
.pio/libdeps/teensy41/TeensyTimerTool/src/types.h:9:10: warning: #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks" [-Wcpp]
         #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks"
          ^
In file included from .pio/libdeps/teensy41/TeensyTimerTool/src/ITimerChannel.h:3:0,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/TimerModules/PIT4/PITChannel.h:4,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/TimerModules/PIT4/PIT.h:3,
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/TimerModules/PIT4/PIT.cpp:3:
.pio/libdeps/teensy41/TeensyTimerTool/src/types.h:9:10: warning: #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks" [-Wcpp]
[... 29 lines cut ...]
                 from .pio/libdeps/teensy41/TeensyTimerTool/src/config.cpp:2:
.pio/libdeps/teensy41/TeensyTimerTool/src/TimerModules/TCK/../../types.h:9:10: warning: #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks" [-Wcpp]
Compiling .pio/build/teensy41/FrameworkArduino/HardwareSerial1.cpp.o
         #warning "can't use staticFunctional with GCC < V7! Fallback to function pointer callbacks"
          ^
In file included from src/with_library.cpp:1:0:
include/PulseGenerator.h: In member function 'void PulseGenerator::begin(const unsigned int&)':
include/PulseGenerator.h:27:46: error: no matching function for call to 'TeensyTimerTool::OneShotTimer::begin(PulseGenerator::begin(const unsigned int&)::<lambda()>)'
   timer_.begin([this]() { this->callback(); });
                                              ^
In file included from .pio/libdeps/teensy41/TeensyTimerTool/src/TeensyTimerTool.h:3:0,
                 from include/PulseGenerator.h:4,
                 from src/with_library.cpp:1:
.pio/libdeps/teensy41/TeensyTimerTool/src/API/oneShotTimer.h:26:15: note: candidate: TeensyTimerTool::errorCode TeensyTimerTool::OneShotTimer::begin(TeensyTimerTool::callback_t)
     errorCode OneShotTimer::begin(callback_t callback)
               ^
.pio/libdeps/teensy41/TeensyTimerTool/src/API/oneShotTimer.h:26:15: note:   no known conversion for argument 1 from 'PulseGenerator::begin(const unsigned int&)::<lambda()>' to 'TeensyTimerTool::callback_t {aka void (*)()}'
Compiling .pio/build/teensy41/FrameworkArduino/HardwareSerial2.cpp.o
*** [.pio/build/teensy41/src/with_library.cpp.o] Error 1
================================================================================ [FAILED] Took 0.72 seconds ================================================================================
```

Note this is w/ 
* `gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1)`
* platformIO

luni64, 2022-11-23:
Sorry, I released v1.1.0 too early. The new callback system requires the current beta version of Teensyduino. In case you do not want to install the beta you need to downgrade to v1.0.0 of the TimerTool. That should work with the currently released Teensyduino. Feature-wise there is no difference between those versions.

mnissov, 2022-11-25:
That seems to have resolved the problem, thanks

closed 2022-11-25 by mnissov