luni64/TeensyTimerTool #26: Support for RTC timer

Defragster, 2023-07-06:
Recent forum posts made use of RTC clock interrupt. For perhaps lower freq interrupt periods it could free up other timers.

Running an example that completes in setup() the T_4.0 went to empty loop and I touched it and it was quite warm to touch and internal reported '60C==140F'.

"wfi" doesn't work without a real interrupt [millis_tick is not 'real'] - and adding an intervaltimer with "WFI" dropped it to 45C==113F
[https://github.com/Defragster/TeensySketches/blob/main/WFIon1062/WFIon1062.ino](url)

Redid the same with RTC _ISR() from recent forum sketch and it worked with no loss in accuracy to the PPS sketch
[https://github.com/Defragster/TeensySketches/blob/main/WFIon1062_RTC/WFIon1062_RTC.ino](url)

Also interesting the PPS sketch showed about 60K clock ticks per half second using WFI instead of 300M - so it really does sleep the processor.

Adding RTC to TimerTool might have value - the rest of this would probably be good for TeensyUser/Wiki ref notes?

luni64, 2023-07-07:
The TimerTool already has RTC timers. Here how to use them:

```c++
#include "TeensyTimerTool.h"
using namespace TeensyTimerTool;

PeriodicTimer myRTCtimer(TCK_RTC); 

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
    myRTCtimer.begin([]{digitalToggleFast(LED_BUILTIN);}, 150ms);
}

void loop()
{
}
```
Does that work in your examples?

Defragster, 2023-07-07:
Odd about the links as the 'literally' work if copied - but entered as URL's they link relative to this repository ...

I'll look at the TCK_RTC version to see it. I did see other WIKI uses of the RTC after I wrote this.

I added a sketch https://github.com/Defragster/TeensySketches/tree/main/PIF_1062_RTC where I made the 4 bit setting a passed parameter to find out the timing of the divisors of the 32K clock:
	setupRTCInterrupt PIF set to 15 0xF
599981092 RTC ISR cycles	999968 us diff	1 Hz
	setupRTCInterrupt PIF set to 14 0xE
299990516 RTC ISR cycles	499984 us diff	2 Hz
	setupRTCInterrupt PIF set to 13 0xD
149995256 RTC ISR cycles	249992 us diff	4 Hz
	setupRTCInterrupt PIF set to 12 0xC
74997628 RTC ISR cycles	124996 us diff	8 Hz
	setupRTCInterrupt PIF set to 11 0xB
37498816 RTC ISR cycles	62498 us diff	16 Hz
	setupRTCInterrupt PIF set to 10 0xA
18749404 RTC ISR cycles	31249 us diff	32 Hz
	setupRTCInterrupt PIF set to 9 0x9
9374704 RTC ISR cycles	15625 us diff	64 Hz
	setupRTCInterrupt PIF set to 8 0x8
4687352 RTC ISR cycles	7812 us diff	128 Hz
	setupRTCInterrupt PIF set to 7 0x7
2343676 RTC ISR cycles	3906 us diff	256 Hz
	setupRTCInterrupt PIF set to 6 0x6
1171840 RTC ISR cycles	1953 us diff	512 Hz
	setupRTCInterrupt PIF set to 5 0x5
585920 RTC ISR cycles	976 us diff	1024 Hz
	setupRTCInterrupt PIF set to 4 0x4
292960 RTC ISR cycles	488 us diff	2048 Hz
	setupRTCInterrupt PIF set to 3 0x3
146480 RTC ISR cycles	244 us diff	4096 Hz
	setupRTCInterrupt PIF set to 2 0x2
73240 RTC ISR cycles	122 us diff	8192 Hz
	setupRTCInterrupt PIF set to 1 0x1
36620 RTC ISR cycles	61 us diff	16384 Hz
	setupRTCInterrupt PIF set to 0 0x0
18308 RTC ISR cycles	30 us diff	32772 Hz

Defragster, 2023-07-07:
Looking at TimerTool sources I don't see these in use: IRQ_SNVS_IRQ or SNVS_HPCR

Not sure how that would be done or exposed to the user - but it could be an option for the 64 bit clock to avoid missing the ARM_DWT_CYCCNT rollover or if a user wanted a callback of 1, 2, 4, ... 16K, 32K Hz. It is only a single RTC timer interrupt - but it could free up another timer as noted giving some utility.

With WFI either intervalTimer or RTC_timer below seemed to work - the RTC might be lower power?

For an _ISR() callback like done here - that gave the 16 values above - this forum example uses PRI 16 - but 255 could often work:
`// sets up the RTC to generate interrupt.
// page 1248: 7-4 PI_FREQ : Periodic Interrupt Frequency values 0-15
void setupRTCInterrupt( uint8_t valPIF ) {
  static bool cbSet = false;
  cntFreq = 0;
  Serial.printf( "\n\tsetupRTCInterrupt PIF set to %u 0x%X\n", valPIF, valPIF );
  // disable periodic interrupt
  SNVS_HPCR &= ~SNVS_HPCR_PI_EN;
  while ((SNVS_HPCR & SNVS_HPCR_PI_EN)) {}  //The RTC is a slow peripheral, need to spin until PI_EN is reset...
  SNVS_HPCR = SNVS_HPCR_PI_FREQ( valPIF );  // change to 0b1111 if you need a 1s interrupt
  // enable periodic interrupt
  SNVS_HPCR |= SNVS_HPCR_PI_EN;
  while (!(SNVS_HPCR & SNVS_HPCR_PI_EN)) {}  // spin until PI_EN is set...

  // attach a callback - doesn't change after first call
  if (!cbSet) {
    cbSet = true;
    attachInterruptVector(IRQ_SNVS_IRQ, callback);
    NVIC_SET_PRIORITY(IRQ_SNVS_IRQ, 16);  // some priority 0-255 (low number = high priority)
    NVIC_ENABLE_IRQ(IRQ_SNVS_IRQ);
  }
}`

Defragster, 2023-07-08:
Pasted code above with edit for "WFI" and it does not work.

`#include "TeensyTimerTool.h"
using namespace TeensyTimerTool;

PeriodicTimer myRTCtimer(TCK_RTC); 

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
    myRTCtimer.begin([]{digitalToggleFast(LED_BUILTIN);}, 500ms);
}

void loop()
{
  asm volatile( "wfi" ); // LOW POWER STALL
}`

This code does work as it uses the RTC Interrupt - and only a REAL interrupt will wake from WFI.
This STOPS the CPU and Cycle Counter and wakes by the RTC - as was in that recent forum thread.
The bad thing is Time is lost with this as no millis_systick events (not a real interrupt) occur to wake the CPU.
Challenge would be to ref the RTC time and update the system millis() and micros() or use it to feed the HiResClock?
Other interrupts will wake from WFI, for instance Serial.print and then immediate WFI will wake as the USB output is processed.
Those waking interrupts would NOT use this callback - so they would not know to check the RTC for lost time.
As noted - I did this when I found my T_4.0 in empty loop() HOT to touch at 60°C and adding the WFI dropped it to 45°C {see [WFIon1062.ino](https://github.com/Defragster/TeensySketches/blob/main/WFIon1062/WFIon1062.ino) above}
`void callback() {
  SNVS_HPSR |= 0b11;               // reset interrupt flag
  digitalToggleFast(LED_BUILTIN);
  asm("dsb");                      // wait until flag is synced over the busses to prevent double calls of the isr
}

void setupRTCInterrupt( uint8_t valPIF ) {
  // disable periodic interrupt
  SNVS_HPCR &= ~SNVS_HPCR_PI_EN;
  while ((SNVS_HPCR & SNVS_HPCR_PI_EN)) {}  //The RTC is a slow peripheral, need to spin until PI_EN is reset...
  SNVS_HPCR = SNVS_HPCR_PI_FREQ( valPIF );  // change to 0b1111 if you need a 1s interrupt
  // enable periodic interrupt
  SNVS_HPCR |= SNVS_HPCR_PI_EN;
  while (!(SNVS_HPCR & SNVS_HPCR_PI_EN)) {}  // spin until PI_EN is set...
  attachInterruptVector(IRQ_SNVS_IRQ, callback);
  NVIC_SET_PRIORITY(IRQ_SNVS_IRQ, 16);  // some priority 0-255 (low number = high priority)
  NVIC_ENABLE_IRQ(IRQ_SNVS_IRQ);
}

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  setupRTCInterrupt( 14 ); // 14 is 2 Hz
}

void loop()
{
  asm volatile( "wfi" ); // LOW POWER STALL
}`

Defragster, 2023-07-08:
Went ahead and posted the [above simple sample](https://github.com/Defragster/TeensySketches/blob/main/WFIon1062_Simple/WFIon1062_Simple.ino)

Also learned github URL links are diff than on forum.pjrc

luni64, 2023-07-18:
Sorry for the late answer. 
I first thought you only need a callback which would be provided by the TCK_RTC timers but they are software based and don't generate "real" interrupt.  I do understand now what you need. 

I'm not sure if this timer fits into the generic TimerTool API but I will have a closer look. Would be a nice addition indeed.

Defragster, 2023-07-18:
Glad it makes sense. It may not make fit in generic TimerTool - but does offer an alternate timer option, and one that helps in the case where "wfi" might get used. I didn't measure the current draw - but it sure drops the temperature with the "wfi" and gives a reliable way to wake up. The IntervalTimer works as well - but that may not 'stop' the processor the same way?  I didn't measure that.

'wfi' really does mess with timer sys_ticks millis/micros of course. My quick effort to wake and sync those didn't turn out very well before I got distracted.

luni64, 2023-07-24:
Added a RTC module to the TimerTool (Branch RTC_Timer, needs TD159beta). The module uses the RTC interrupts which can generate interrupts with periods of  1s, 0.5s, 0.25s. .... 30.5µs. If you request a period which the RTC doesn't support, the next longer allowed period is used. 
The interrupts are synced to the RTC. E.g. if you do a 1s periodic timer the callback will be called in sync with the 1s transitions of the RTC.
 
Here a quick usage example: 

```c++
#include "Arduino.h"
#include "TeensyTimerTool.h"
using namespace TeensyTimerTool;

PeriodicTimer myRTCtimer(RTC);
//OneShotTimer myRTCtimer(RTC);

void setup()
{
    while (!Serial) {}

    TeensyTimerTool::attachErrFunc(TeensyTimerTool::ErrorHandler(Serial)); // print library error messages to serial 

    pinMode(LED_BUILTIN, OUTPUT);
    myRTCtimer.begin([] { digitalToggleFast(LED_BUILTIN); }, 25ms);

    delay(2000);
    myRTCtimer.stop();
    delay(2000);
    myRTCtimer.setPeriod(0.5s);
    delay(5000);
    myRTCtimer.setPeriod(50ms);
}
  
void loop()
{
}
```

luni64, 2023-07-26:
Included in v1.4.0

closed 2023-07-26 by luni64