luni64/TeensyTimerTool #29: error: expected '{' before '::' token

bblanchon, 2023-08-23:
Hi @luni64,

I'm having the following error when compiling with arduino-cli:

```text
In file included from /home/runner/Arduino/libraries/TeensyTimerTool/src/TimerModules/RTC/RTC.cpp:3:0:
/home/runner/Arduino/libraries/TeensyTimerTool/src/TimerModules/RTC/RTC.h:[5](https://github.com/bblanchon/ArduinoContinuousStepper/actions/runs/5952260840/job/16143840558#step:10:6):2[6](https://github.com/bblanchon/ArduinoContinuousStepper/actions/runs/5952260840/job/16143840558#step:10:7): error: expected '{' before '::' token
 namespace TeensyTimerTool::RTC_t
                          ^
/home/runner/Arduino/libraries/TeensyTimerTool/src/TimerModules/RTC/RTC.h:5:2[8](https://github.com/bblanchon/ArduinoContinuousStepper/actions/runs/5952260840/job/16143840558#step:10:9): error: 'RTC_t' in namespace '::' does not name a type
 namespace TeensyTimerTool::RTC_t
                            ^
/home/runner/Arduino/libraries/TeensyTimerTool/src/TimerModules/RTC/RTC.cpp:6:26: error: expected '{' before '::' token
 namespace TeensyTimerTool::RTC_t
                          ^
/home/runner/Arduino/libraries/TeensyTimerTool/src/TimerModules/RTC/RTC.cpp:6:28: error: 'RTC_t' in namespace '::' does not name a type
 namespace TeensyTimerTool::RTC_t
                            ^
/home/runner/Arduino/libraries/TeensyTimerTool/src/TimerModules/RTC/RTC.cpp:32:1: error: expected '}' at end of input
 } // namespace TeensyTimerTool::RTC_t
 ^
/home/runner/Arduino/libraries/TeensyTimerTool/src/TimerModules/RTC/RTC.cpp:32:1: error: expected '}' at end of input

Error during build: exit status 1
```

Nested namespace definitions require C++17; I think it's a bit too early for that.

Best regards,
Benoit

dzalf, 2023-10-31:
Dear @luni64 

Same issue here!

The code used to work fine some months ago. Now, my project is broken and won't compile:

[image]

as @bblanchon mentioned is this related to C++17 features?

dzalf, 2023-10-31:
Guy, I think I spoke too fast. 😞 

I was using an old version of the library. After updating it to 1.4.1 in my `platformio` project, the code compiles fine

`luni64/TeensyTimerTool@^1.4.1` yields:

[image]

us866, 2023-11-04:
luni64 is right might be broke:  none of the github examples compile: all give this same error.
TeensyTimerTool\src\TimerModules/RTC/RTC.h:5:26: error: expected '{' before '::' token
 namespace TeensyTimerTool::RTC_t

us866, 2023-11-04:
tried 1.4.1 and 1.4.0 library

luni64, 2023-11-09:
Please see here: https://forum.pjrc.com/index.php?threads/teensy-timer-tools-error.73494/post-331340